adafruit/Adafruit-Retrogame #17: Button repeating but not held.

skpacman, 2015-03-31:
Does it recognize button press on Rising edge and release on falling edge, or what? I have my buttons pulled down then when the button is pressed it reads high. Is this opposite of how retrogame works?

I can't rewire it because the circuit is set up in such a way as when any button is pressed, a LED lights inside the game cabinet.

Can retrogame be changed so it recognizes the button press at a HIGH signal instead of LOW?

skpacman, 2015-04-01:
I just went through the code last night and found what's going on. Retrogame is pulling the inputs high and reading them as active_low.

Is there any way to reverse this logic? Can I have it start low and read active_high??

PaintYourDragon, 2015-04-01:
Around line 330:
        gpio[GPPUD]     = 2;                    // Enable pullup
Change the value to '1' to enable pull-down (instead of pull-up).
You'll then need to look through the code for references to the intstate[] array -- reverse the '0' and '1' checks on the 'c' variable following the read() call.

closed 2015-04-01 by PaintYourDragon

skpacman, 2015-04-01:
Thanks. I'll look through the code tonight and let you know how it goes.

skpacman, 2015-04-03:
Changed the code as described and works great. I was going to use some logic inverters for the circuit if it was too hard to change the code, but it looks like I won't be needing them.

skpacman, 2016-12-02:
Sorry to necromance an old closed issue but:
My arcade recently needed an overhaul and I updated retrogame to the latest version. How can I achieve the same effect with the new version?

PaintYourDragon, 2016-12-02:
The code has gotten rather hairy as of late in order to support I2C port expanders. Trying to make the same changes is the path to madness.

If the old version did what you needed, I'd suggest sticking with it. There've been no performance improvements, just added features for some current and future projects.

If you didn't back that up, click the "commits" link (above the repository file list) and locate a version of the code of similar vintage, then make the described changes.

skpacman, 2016-12-03:
I do have a backup of my previous installation, I'll just stick with that, then. Thanks for the quick reply!