adafruit/Adafruit-Retrogame #18: retrogame + PiTFT + Archlinux

shenron, 2015-05-15:
Hi,

I want to use the project PIGRRL with an Archlinux.

For the screen PITFT, I add these lines in config.txt:
> device_tree_overlay=pitft28
> dtoverlay= pitft28,rotate=270,speed=32000000,fps=60

And I add pitft28-overlay.dt bin /boot/overlays/

For the retrogame (keyboard map GPIO): 
I have a raspberry pi b rev1 (256 ram), so I must update map.
I change the pin numbers 2, 3 and 27 to 0, 1 and 21, respectively. Recompile with:
> make CFLAGS=-DCUPCADE
> sudo make install
> modprobe uinput
> sudo retrogame

But I have this error:
> ./retrogame: Pin config failed.  Try 'sudo ./retrogame'.

Have you an idea ?

Thank you.

mmaret, 2015-05-18:
May be you have some GPIO already used by something else ?
You can modify the `err("Pin config failed");` in retrogame.c to `printf("Cannot config pin %d\n",io[i].pin);
err("Pin config failed");` to know which pin is already used.

nitnoy, 2016-01-01:
@mmaret Thanks for that!

d3rio, 2016-08-04:
I'm trying to set up the retrogame joystick for using in MAME as well.  My setup is almost exactly the same as the project page [here](https://learn.adafruit.com/retro-gaming-with-raspberry-pi?view=all), the only difference is that I have slightly changed which GPIO pins I'm using since I'm using a breakout board to go from the GPIO to a breadboard and then wiring the joystick and buttons into the breadboard, and I can use the ground rail so that I only need one of the ground GPIO pins connected to the ground rail, and then wire the joystick and the button grounds all to the same ground rail.

Instead of the wiring shown in the project page, I just started at the upper left and used the first 6 GPIO pins: 2, 3, 4, 17, 27, and 22, with the GND pin between 4 and 17 connected to the ground rail.   At line 117 in retrogame.c I changed it to the following:
`#if 0`
`        // From our original Pi gaming guide:`
`        // Input   Output (from /usr/include/linux/input.h)`
`        {   2,     KEY_LEFT     },   // Joystick (4 pins)`
`        {   3,     KEY_RIGHT    },`
`        {   4,     KEY_UP       },`
`        {  17,     KEY_DOWN     },`
`        {  27,     KEY_LEFTCTRL },   // A/Fire/jump/primary`
`        {  22,     KEY_LEFTALT  },   // B/Bomb/secondary`

Then I saved and exited, and used the makefile with the command
`make retrogame`
For output I get
`gcc  -Wall -Ofast -fomit-frame-pointer -funroll-loops -s retrogame.c -o retrogame`
echoed to the screen, then after a few seconds I get 
`strip retrogame`
when the compiler finishes.  Then I run it using 
`sudo ./retrogame`
and I get the same error as above:
`./retrogame: Pin config failed.  Try 'sudo ./retrogame'.`

So, after googling the error I found this page and I used the suggestion to modify the `err("Pin config failed");` to `{ printf("Cannot config pin %d\n",io[i].pin);  err("Pin config failed"); }`.  Re-compiling and running it again, I get the following output:
`Cannot config pin 16`
`./retrogame: Pin config failed.  Try 'sudo ./retrogame'.`

Now I have _no_ idea what that means.  There is no GPIO16, and the actual 16th pin the GPIO array is GPIO23, which I'm not using.

Any ideas on what I might be doing wrong?

mmaret-geny, 2016-08-05:
Hello,
If it's starts by #if 0, it will not be used !
That's why it's saying that it "cannot config the pin 16". it's because it's using another config.
Change this #if 0 into #if 1

d3rio, 2016-08-05:
OK, that's something I should have figured out on my own.  I haven't really done any C programming in my life, but I at least do understand that 0 means false and 1 means true.

Changing that the compiled program runs and doesn't seem to give any errors, though I don't know if it works in MAME or anything yet.  When pressing the A button it kills all power then reboots when I let go of the button, this seems to occur even afterwards when retrogame isn't running, so that implies to me that I've wired something incorrectly, though what it is I cannot imagine.  I'll check over everything and report back.

mmaret-geny, 2016-08-08:
How things are going ?

d3rio, 2016-08-10:
Sorry, I've been out of town on business, so I haven't been able to work on it.  I'll follow up once I get home.

d3rio, 2016-08-18:
@mmaret-geny, sorry for the hiatus.  Instead of trying something different, I decided to try doing it exactly like the guide [here](https://learn.adafruit.com/retro-gaming-with-raspberry-pi/buttons?view=all), with the exact same code and pinout.

retrogame.c appears to compile fine without any errors, and when I run it I don't get any errors, it just sits there until I hit ctrl+c to exit, which is how it's supposed to behave if I understand correctly. 

Nothing seems to happen when I move the joystick or press buttons while retrogame is running, but since the arrow keys and ctrl and alt keys don't print anything to the screen anyway, that doesn't tell me anything.

There is one exception though.  Moving the joystick to the RIGHT cuts power to the raspberry pi and causes it to reboot.  I tested this without running retrogame and it *still* does that, which means that this power killing is more fundamental and seemingly has nothing to do with retrogame.  The right channel of the joystick is plugged into GPIO 9, and testing it with a spare wire shorting GPIO 9 to one of the GND pins does kill *all* power to the raspberry pi: even the power LED's on the board go out.

Even weirder, when I measure the voltage potential between GPIO 9 and GND, I get 5V!  It seems that something is fundamentally wrong with my GPIO, but I have no idea what it could be.

PaintYourDragon, 2016-08-18:
@d3rio Are you using this with a PiTFT? A bunch of GPIO pins (including 9) are off-limits in that case:
https://learn.adafruit.com/assets/23709

mmaret-geny, 2016-08-18:
About retrogame PIN definition:
Pin can be either defined in ioTFT[] array or in ioStandard[].
Choosing one over the other is done line 706
`// Select io[] table for Cupcade (TFT) or 'normal' project.`
` io = (access("/etc/modprobe.d/adafruit.conf", F_OK) || `
`access("/dev/fb1", F_OK)) ? ioStandard : ioTFT;`

If the file /etc/modprobe.d/adafruit.conf exist or if a screen is plugged on pin, ioStandard will be used.
If I recall correctly your pin are defined in ioStandard. 
So you can either:
* put the same definition in ioTFT
* make sure to have /etc/modprobe.d/adafruit.conf (you can create it (even empty) with `sudo touch /etc/modprobe.d/adafruit.conf`)

d3rio, 2016-08-18:
@PaintYourDragon  no, I'm not using anything like that.  It's just a ragular Raspberry Pi Model B and I'm using a HDMI monitor for video.

d3rio, 2016-08-19:
Crap... while fiddling with it today I accidentally broke the housing for the SD card slot so I think my old Model B is toast.  I have a Pi 3 Model B as well, but with the different number of pins I'll have to do some homework to figure out the pinout and such.  Does the retrogame.c still work with the newer models of the Raspberry Pi?

PaintYourDragon, 2016-08-19:
retrogame works on Pi 3. If you're using the ready-made PiGRRL 2 SD image though, that only works on Pi 2, not 3.