adafruit/Adafruit-Retrogame #19: Raspberry Pi 2

CapnRobot, 2015-05-17:
Trying to set this up to work with the Pi 2 and some buttons work... but others like GPIO21 (pin40) pressed to the ground next to it, does nothing?

To debug I've mapped 4, 17, 24 ,25 ,5, 6, 12, 13, 19 16, 26, 20, 21
all to different keyboard letters. 

26, 19, 12, 13, 6 and 4 work

20 sends randomly  when i press the other buttons :(

the rest don't work at all. I've checked for shorts everywhere and all the buttons themselves work.
Here's what I'm working on: http://i.imgur.com/ocLdTe5.jpg
And these are the insides: http://i.imgur.com/phMY4Ku.jpg

CapnRobot, 2015-05-17:
Welp, turns out one of the grounds I was using just plain didn't work... XD fixed it.

Gronis, 2015-10-05:
Maybe someone should close this issue? Also, it would be good to add that it works with the raspberry pi in the readme file.

closed 2016-08-11 by PaintYourDragon