adafruit/Adafruit-Retrogame #21: Retrogame requires keyboard plugged in

cincospenguinos, 2015-05-28:
I'm using Retrogame along with Retropie to create a portable gaming system for my little brother. Right now I'm using Retrogame to listen for inputs through the GPIO pins on the Raspberry Pi. The program works great, but only if a keyboard is plugged in. I looked around and read lots of blog posts and forums, and I can't find any information concerning a work around. Is there a way for retrogame to work without a keyboard plugged in?

Thank you for your help.

Plaputta, 2015-12-14:
You have to add the module "evdev"
`sudo nano /etc/modules`
...then add the line
`evdev`
Reboot and it'll work.

fmaurer, 2016-07-20:
I wasn't able to get the evdev workaround mentioned above to work. On fresh Raspbian Jessie install, had to add python tools, then could install evdev:

- added evdev to /etc/modules, then:

```
sudo apt-get install python-dev
sudo pip install evdev
```

However it had no effect, GPIO buttons still don't work unless keyboard present at boot. Any suggestions? Can the changes to /etc/udev/rules.d/10-retrogame.rules interfere with it somehow?

rofltech, 2016-08-10:
I have the same problem.  evdev and uinput are both looaded when  check with lsmod.  When keyboard is present t works soon as i unplug the keyboard everything stops.

PaintYourDragon, 2016-08-11:
Think this is fixed now, give it a try!

rofltech, 2016-08-11:
Thanks PaintYourDragon.  You can mark this closed as far as I'm concerned it works now.  Not sure if OP retried

closed 2016-10-17 by ladyada