adafruit/Adafruit-Retrogame #24: Pi freezes on activation of retrogame

spots1000, 2016-01-15:
Hi, so my pi is running the adafruit issue lastest version of rasbian with a ptfit display attached. I am running Retrogame to get button inputs set up for Retropi, and everthing works fine until I actualy start retrogame. Whenever I start retrogame, by going to /usr/local/bin and doing sudo ./retrogame, it just goes to the next line and freezes entirely. I compiled it multiple times, but it continues to do this even though it compiled fine, and the pi even tried to activate it on startup a few times, causing a freeze than too. After it freezes, it locks up completely and excepts no input from either the buttons or a keyboard. The only way to unfreeze it is to hard shutdown and reboot the pi.

PaintYourDragon, 2016-01-15:
Shoot. Raspbian Jessie? Which model of Pi board exactly? I'm away from Pi for a couple days, but will take a look at this when I get back.

spots1000, 2016-01-15:
It's a brand new pi 1 b+ board. Everything works fine except retrogame, even the ptfit display.

spots1000, 2016-02-12:
Any progress?

mmaret-geny, 2016-02-12:
If you run cat /proc/kmsg in a terminal, an ./retrogame in another one (let's say ssh). Did you have any message before it freeze ?

If you have some C knowledge, may be you can add some debug information like:
line 355 add the following line:
printf("Configuring pin[%d]= %d",i, io[i].pin);
recompile, re-run and give us the output.

If the pin are not the one you expect, make sure you have the following file available(event empty) :/etc/modprobe.d/adafruit.conf

The last, thing you can try, is to past the result of sudo strace ./retrogame

PaintYourDragon, 2016-02-12:
Would be helpful to know what's in your pin table. Perhaps there's some conflict with another function on that pin.

spots1000, 2016-02-12:
How could I check that? It is quite possible.

PaintYourDragon, 2016-02-14:
Copy-and-paste the contents of the table here. Either the ioStandard[] table if you're using HDMI or composite video out, or ioTFT[] if using a PiTFT screen.

spots1000, 2016-02-14:
Where would I get the table from, I found the unedited one but I can't find
the one that I compiled with.

bbmech7, 2016-09-15:
im haveing the same issues even with the stock file. using pizero with pitft. i ran  with strace last and this is the output

`sudo strace ./retrogame
execve("./retrogame", ["./retrogame"], [/* 14 vars */]) = 0
brk(0)                                  = 0x14ae000
uname({sys="Linux", node="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f2b000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6f2a000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\5\0\0004\0\0\0"..., 512) = 512
lseek(3, 17960, SEEK_SET)               = 17960
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960) = 960
lseek(3, 17696, SEEK_SET)               = 17696
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0644, st_size=18920, ...}) = 0
mmap2(NULL, 83236, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ee9000
mprotect(0xb6eee000, 61440, PROT_NONE)  = 0
mmap2(0xb6efd000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0xb6efd000
mprotect(0xbec96000, 4096, PROT_READ|PROT_WRITE|PROT_EXEC|PROT_GROWSDOWN) = 0
close(3)                                = 0
munmap(0xb6f2a000, 42)                  = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=80856, ...}) = 0
mmap2(NULL, 80856, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6ed5000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0L\204\1\0004\0\0\0"..., 512) = 512
lseek(3, 1239936, SEEK_SET)             = 1239936
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 2840) = 2840
lseek(3, 1236500, SEEK_SET)             = 1236500
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0755, st_size=1242776, ...}) = 0
mmap2(NULL, 1312152, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6d94000
mprotect(0xb6ebf000, 65536, PROT_NONE)  = 0
mmap2(0xb6ecf000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12b000) = 0xb6ecf000
mmap2(0xb6ed2000, 9624, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ed2000
close(3)                                = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f2a000
set_tls(0xb6f2a4c0, 0xb6f2aba8, 0xb6f2e050, 0xb6f2a4c0, 0xb6f2e050) = 0
mprotect(0xb6ecf000, 8192, PROT_READ)   = 0
mprotect(0xb6ee9000, 20480, PROT_READ|PROT_WRITE) = 0
mprotect(0xb6ee9000, 20480, PROT_READ|PROT_EXEC) = 0
cacheflush(0xb6ee9000, 0xb6eee000, 0, 0x15, 0xbec963c0) = 0
mprotect(0xb6f2d000, 4096, PROT_READ)   = 0
munmap(0xb6ed5000, 80856)               = 0
brk(0)                                  = 0x14ae000
brk(0x14cf000)                          = 0x14cf000
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1], NULL, 8) = 0
signalfd4(-1, ~[RTMIN RT_1], 8, 0)      = 3
inotify_init()                          = 4
inotify_init()                          = 5
inotify_add_watch(4, "/boot/retrogame.cfg", IN_MODIFY|IN_IGNORED) = -1 ENOENT (No such file or directory)
inotify_add_watch(5, "/boot", IN_MOVE|IN_CREATE) = 1
open("/proc/cmdline", O_RDONLY)         = 6
fstat64(6, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f29000
read(6, "dma.dmachans=0x7f35 bcm2708_fb.f"..., 1024) = 394
read(6, "", 1024)                       = 0
close(6)                                = 0
munmap(0xb6f29000, 4096)                = 0
open("/dev/mem", O_RDWR|O_SYNC)         = 6
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_SHARED, 6, 0x20200000) = 0xb6f29000
close(6)                                = 0
access("/etc/modprobe.d/adafruit.conf", F_OK) = 0
[... 62 lines cut ...]
write(6, "22", 2)                       = -1 EBUSY (Device or resource busy)
open("/sys/class/gpio/gpio22/active_low", O_WRONLY) = -1 ENOENT (No such file or directory)
open("/sys/class/gpio/gpio22/direction", O_WRONLY) = -1 ENOENT (No such file or directory)
fstat64(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f28000
write(1, "retrogame: Pin config failed.  T"..., 55retrogame: Pin config failed.  Try 'sudo ./retrogame'.
) = 55
close(7)                                = 0
close(8)                                = 0
close(9)                                = 0
close(10)                               = 0
close(11)                               = 0
open("/sys/class/gpio/unexport", O_WRONLY) = 7
write(7, "2", 1)                        = 1
write(7, "3", 1)                        = 1
write(7, "4", 1)                        = 1
write(7, "17", 2)                       = 2
write(7, "27", 2)                       = 2
write(7, "22", 2)                       = -1 EINVAL (Invalid argument)
write(7, "23", 2)                       = -1 EINVAL (Invalid argument)
write(7, "18", 2)                       = -1 EINVAL (Invalid argument)
close(7)                                = 0
exit_group(1)                           = ?
+++ exited with 1 +++
`

mmaret, 2016-09-15:
>
> im haveing the same issues even with the stock file. using pizero with
pitft. i ran with strace last and this is the output
> sudo strace ./retrogame
> execve("./retrogame", ["./retrogame"], [/* 14 vars */]) = 0
> brk(0) = 0x14ae000
> uname({sys="Linux", node="raspberrypi", ...}) = 0
> access("/etc/ld.so.nohwcap", F_OK) = -1 [...]
= 0xb6f2b000
> access("/etc/ld.so.preload", R_OK) = 0
> open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
> fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
> mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, [...]
"\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h\5\0\0004\0\0\0"..., 512) =
512
> lseek(3, 17960, SEEK_SET) = 17960
> read(3,
"\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 960)
= 960
> lseek(3, 17696, SEEK_SET) = 17696
> read(3,
"A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 47) = 47
> fstat64(3, {st_mode=S_IFREG|0644, st_size=18920, ...}) = 0
> mmap2(NULL, 83236, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0)
= 0xb6ee9000
> mprotect(0xb6eee000, 61440, PROT_NONE) = 0
> mmap2(0xb6efd000, 4096, PROT_READ|PROT_WRITE,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0xb6efd000
> mprotect(0xbec96000, 4096, PROT_READ|PROT_WRITE|PROT_EXEC|PROT_GROWSDOWN)
= 0
> close(3) = 0
> munmap(0xb6f2a000, 42) = 0
> open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
> fstat64(3, {st_mode=S_IFREG|0644, st_size=80856, ...}) = 0
> mmap2(NULL, 80856, PROT_READ, MAP_PRIVATE, 3, 0) = [...]
"\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0L\204\1\0004\0\0\0"..., 512)
= 512
> lseek(3, 1239936, SEEK_SET) = 1239936
> read(3,
"\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"...,
2840) = 2840
> lseek(3, 1236500, SEEK_SET) = 1236500
> read(3,
"A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
> fstat64(3, {st_mode=S_IFREG|0755, st_size=1242776, ...}) = 0
> mmap2(NULL, 1312152, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3,
0) = 0xb6d94000
> mprotect(0xb6ebf000, 65536, PROT_NONE) = 0
> mmap2(0xb6ecf000, 12288, PROT_READ|PROT_WRITE,
MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12b000) = 0xb6ecf000
> mmap2(0xb6ed2000, 9624, PROT_READ|PROT_WRITE,
MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ed2000
> close(3) = 0
> mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0)
= 0xb6f2a000
> set_tls(0xb6f2a4c0, 0xb6f2aba8, 0xb6f2e050, 0xb6f2a4c0, 0xb6f2e050) = 0
> mprotect(0xb6ecf000, 8192, PROT_READ) = 0
> mprotect(0xb6ee9000, 20480, PROT_READ|PROT_WRITE) = 0
> mprotect(0xb6ee9000, 20480, [...]
ENOENT (No such file or directory)
> inotify_add_watch(5, "/boot", IN_MOVE|IN_CREATE) = 1
> open("/proc/cmdline", O_RDONLY) = 6
> fstat64(6, {st_mode=S_IFREG|0444, st_size=0, ...}) = 0
> mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0)
= 0xb6f29000
> read(6, "dma.dmachans=0x7f35 bcm2708_fb.f"..., 1024) = 394
> read(6, "", 1024) = 0
> close(6) = 0
> munmap(0xb6f29000, 4096) = 0
> open("/dev/mem", O_RDWR|O_SYNC) = 6
> mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_SHARED, 6, 0x20200000) =
0xb6f29000
> close(6) = 0
> access("/etc/modprobe.d/adafruit.conf", F_OK) = 0
> access("/dev/fb1", F_OK) = 0
> open("/sys/class/gpio/export", O_WRONLY) = 6
> write(6, "2", 1) = 1
> open("/sys/class/gpio/gpio2/active_low", [...]
file or directory)
> open("/sys/class/gpio/gpio22/direction", O_WRONLY) = -1 ENOENT (No such
file or directory)

Looks like pin 22 is already in use.
> fstat64(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
> mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0)
= 0xb6f28000
> write(1, "retrogame: Pin config failed. T"..., 55retrogame: Pin config
failed. Try 'sudo ./retrogame'.

PaintYourDragon, 2016-09-15:
Can't reproduce, need more information. Exact OS image & version, any & all hardware attached to the header, and any other GPIO-related code that might be running on the system.

bbmech7, 2016-09-16:
I'll have to check it out when i get home. So far it was only the pitft. I
even went  and commented out most of the button defs in that area and made
sure that the three pins with errors were not being defined.... on second
thought they may have been used in the Vulcan nerve pinch area... ill have
too check more sorry.