adafruit/Adafruit-Retrogame #25: Some inputs will not respond at random

Qjs, 2016-02-08:
Hi there!

I built a nice little arcade box with an RPi2, 6 Arcade buttons, and an adafruit Small Joystick. Everything is wired up fine, and responds accordingly most of the time. However after some time (at random) I notice that a certain input wont respond. I am using MAME as my arcade platform for now. For example, I'll be playing a game then all of the sudden the Up motion on the joystick won't respond, or the Shoot arcade button will just not react. After a reboot everything will work for awhile again, then the error will happen. I am puzzled to believe this is still a hardware issue, have there been any known issues with MAME using retrogame? I have made sure SDL2 compatibility is okay and everything.

PaintYourDragon, 2016-02-08:
Haven't heard reports of this before...any controls working at startup should stay 'live' as long as the system's running. Are you using a ready-made image such as RetroPie, or did you build this up yourself? If the latter, exactly what OS & release was the starting point? I'll keep an eye out for any other reports of the same.

CrazySpence, 2016-02-08:
Check your grounds and connections, if they're loose or something shorted across between the pins it'll cause that behaviour

closed 2016-10-17 by ladyada