adafruit/Adafruit-Retrogame #27: Inputs not responding when using GPIO arcade buttons on Maldita Castilla (port game) in RetroPie

hiulit, 2016-08-11:
Hi there,
First of all, I don't know if this issue would fit here, so if not, feel free to close it.

I've installed a port by [YoYo Games](http://www.yoyogames.com/pi) of the amazing game **Maldita Castilla** for the Raspberry Pi.

I've followed [these instructions](https://github.com/bite-your-idols/gamemaker-retropie) and it's working perfectly, but only when using a keyboard, not when pressing the GPIO arcade buttons.

Do you know why that might be?
The other ports that come bundled with RetroPie work flawlessly. And also all the emulators.

Thanks!

PaintYourDragon, 2016-08-17:
See the retrogame README for directions on setting up a udev rule, see if that helps.

hiulit, 2016-08-17:
Thanks!
I think I already did that, but I'll check again and I'll let you know.
The thing is that's only happening with that particular game I mentioned in the other issue.

hiulit, 2017-03-21:
HI there, it's been a while.
I set up udev rule as following instructions and it still doesn't work

PaintYourDragon, 2017-03-21:
There are multiple ways of both generating and capturing keyboard events in Linux. It appears this game and retrogame are using two mutually incompatible methods, and without information on how the former is operating, can't resolve this.

retrogame was designed to be paired with RetroPie. If it happens to work with other things, that's "cake," but if not, it's not something I can scramble to fix.

I'd suggest trying out the latest retrogame (the udev stuff was cleaned up a little) and making a configuration file to match the game's keyboard input...but if that doesn't work, only option then is connecting a USB keyboard when you want to use this game.

closed 2017-03-21 by PaintYourDragon