adafruit/Adafruit-Retrogame #28: GPIO work in retropie but not in retroarch

jefflequeux, 2016-08-16:
I use GPIO with this config:

	{  22,     KEY_LEFT     }, // Joystick (4 pins)
	{  23,     KEY_RIGHT    },
	{  17,     KEY_UP       },
	{  27,     KEY_DOWN     },
	{  24,     KEY_LEFTCTRL }, // A/Fire/jump/primary/RED
	{  10,     KEY_LEFTALT  }, // B/Bomb/secondary/YELLOW
	{   9,     KEY_Z        }, // X/BLUE
	{  25,     KEY_X        }, // Y/GREEN
	{  18,     KEY_SPACE    }, // Select
	{   4,     KEY_ENTER    }, // Start
	{  11,     KEY_A        }, // L Shoulder
 	{   8,     KEY_S        }, // R Shoulder
	{  13,     KEY_1        }, // PiTFT Button 2
	{  19,     KEY_2        }, // PiTFT Button 3
	{  26,     KEY_3        }, // PiTFT Button 4

It works great on retropie interface but not in game, retroarch doesn't recognize this GPIO, doesn't know why.

hiulit, 2016-08-16:
Maybe this issue is somehow related to [this issue](https://github.com/adafruit/Adafruit-Retrogame/issues/27)?

But in my case, I can play all the games in all the emulators except for the one in the my issue.

PaintYourDragon, 2016-08-17:
Please see the README for this repository for directions on setting up a UDEV rule; this should make retroarch recognize these inputs. You may need to further edit /opt/retropie/configs/all/retroarch.cfg to match the keys you've set up (the config isn't carried over from emulationstation for some reason).

closed 2016-08-17 by PaintYourDragon