adafruit/Adafruit-Retrogame #29: Raspberry Pi 3 Model B

mcallistertyler95, 2016-08-17:
Is this compatible with Raspberry Pi 3 Model B? Not getting any response from my buttons and as far as I am aware the GPIO is the same as the 2 models.

jlowe64, 2016-08-17:
I have a similar issue, but I'm getting a response out of `up`, `left`, left and right bumpers, but that's it. The `Up` is like `B`, the `left` is like `A`, `left bumper` is `left`, and `right bumper` is `down`.

PaintYourDragon, 2016-08-18:
Very definitely 100% Raspberry Pi 3 compatible; that's been my development system lately.

Non-responsiveness and/or unexpected inputs are almost certainly the Retroarch configuration file. input_driver and input_joypad_driver should both be set to "udev", along with the udev directions on the repository home page. You'll also need to set up the directions and A/B/etc buttons, for whatever reason these don't transfer from EmulationStation to all of the Retroarch system configs.

Main Retroarch config appears to be /opt/retropie/configs/all/retroarch.cfg
Some system configs have their own key bindings and such that override the defaults, others just include the defaults and call it done. Unfortunately you'll have to go picking through any/all retroarch.cfg files to locate trouble spots.

closed 2016-08-18 by PaintYourDragon

jlowe64, 2016-08-18:
I've just edited `/opt/retropie/configs/all/retroarch.cfg` to reflect the correct mapping. `input_driver` didn't have "udev" so I added that. I've restarted the system and it's still not working. Is there an easier way to debug where the mappings are being pulled from? It's hard to debug the handover between which controllers are being used. I did test going into a game, and most of the keys seemed to work correctly, minus the `a` and `b` keys, but I have a feeling that was the retropie retroarch config issue that I just adjusted. 

Still, the front end for retropie doesn't seem to have the correct mappings. Any help on tracking this down would be appreciated! 

Thank you.

hiulit, 2016-08-18:
You could try the RetroPie forum https://retropie.org.uk/forum See if they can help you out ;)

jlowe64, 2016-08-19:
@hiulit I just posted there as well, and I've seen no response in over 24 hours, so I will assume it'll get buried and never seen. I'm afraid to "bump" because I may get annoying/banned.

jlowe64, 2016-08-19:
@PaintYourDragon Is there any way you can post examples of your config files all the way from EmulationStation, to RetroPie, to RetroArch, etc? Maybe a non-system specific example of each would help narrow down things?

I'm sorry for being annoying, but there's an obvious config issue here and I just haven't found it in any of the files.

PaintYourDragon, 2016-08-19:
Can't do this right now, there's pressing issues elsewhere I'm dealing with, other projects and repositories.

We've discussed maybe re-doing our gaming project SD images (PiGRRL2, PiGRRL Zero, etc.) to support the latest hardware (Pi 3) and maybe clean up some other loose ends. If/when that happens, we'll make sure to have a working config all the way down. But this is all just ideas right now, there's no ETA, could be weeks or months, if ever.

jlowe64, 2016-08-20:
Just so everyone knows. The fix is to delete the emulation station controller config. It'll prompt you to redo your control next time you log in, and then you can use the PiGRRL controller instead.