adafruit/Adafruit-Retrogame #31: GamePad not registering

mikeKane, 2016-09-21:
Just followed https://www.youtube.com/watch?v=4u1NKok2-BE to get the PiTFT to work on a Rasp 3, but when it got to this step I used the new commit where we do not change the ioStandard table at all but when I boot up the Rasp again the input Gamepad is not detected. I checked the voltage and all seems ok. This last commit (67a447ea59626840511fce1b610416065e8788a1) seems to be the only difference in how it works.

PaintYourDragon, 2016-09-21:
Place retrogame.cfg in /boot
Edit as necessary; default config is for PiGRRL 2. Comments in file explain syntax.

PaintYourDragon, 2016-09-21:
Also, it's normal that no gamepad is detected. Retrogame appears as a keyboard...once cfg is set up, you can hold a button and the RetroPie setup will detect it as a keyboard and prompt you through each button.

mikeKane, 2016-09-21:
I have followed the instructions. Below you will find the ls output of boot and the contents of retrogame.cfg and retro rules. 

Still does not register output. One button on the PiTFT does make it reboot however. (Button 2)

[image]
[image]
[image]
[image]

PaintYourDragon, 2016-09-21:
Tested and confirmed working against pristine RetroPie install; Raspberry Pi 3 with PiGRRL 2 gamepad. Things to check:
1. Is retrogame being started in the background on system startup? (e.g. added to /etc/rc.local). Use 'ps' command etc to determine whether it's actually running.
2. Were you previously running an earlier version of retrogame on this system? If so, confirm that it's not the old version being run.
3. Are you using the PiGRRL 2 gamepad? If not, edit /boot/retrogame.cfg to match your GPIO pinout.

mikeKane, 2016-09-21:
Reformatted SD card. Re imaged everything. Changed Power cord. Works flawless! TY

closed 2016-09-21 by mikeKane