adafruit/Adafruit-Retrogame #35: Inputs not working in games.

Joe-does-Calculus, 2016-10-04:
Making a Pigrrl project somewhere between a Pigrrl 2 and Zero. I'm using a Raspberry Pi 3 and have retropie and retrogame installed. Buttons work great in emulationstation but do nothing in a game. Any ideas on what's going wrong?

washplant26, 2016-10-04:
Lucky. I can't even get one single button to do anything.

PaintYourDragon, 2016-10-04:
Probably udev rules related; please see top-level README for setting this up. See if other emulators fare any better (e.g. MAME, NES).
Code is tested and known working with current RetroPie and Pi 3.

closed 2016-10-04 by PaintYourDragon

Joe-does-Calculus, 2016-10-04:
It was a typo in the udev rules, working nicely now. Thank you so much for the help and apologies for misusing the issues tab. I'm still learning.

PaintYourDragon, 2016-10-04:
Cool, glad to hear you got it working.
No worries, took it as an opportunity to clarify in the README where to best find help. And we've been contemplating easier install procedures.