adafruit/Adafruit-Retrogame #40: installing on PI 2 and choosing arcade 6 button , after reboot PI input doesnt work

xillwillx, 2017-02-16:
Had an arcade that used to use the old code that had the indiecity mame install.
raspberry pi 2 using gpio pins with 6 buttons and a joystick wanted to update to retropie 
installed retropie from github on november 2016 raspian image w/ pixel desktop
i installed Retrogame and configured for 6 button + joystick, after reboot keyboard/mouse wont work anymore  and the networking (hardwired) no longer works.

Figuring it might be something i did wrong I tried with a default latest Retropie image this time and it did that same thing after reboot when i pulled down the retrogame.sh and rebooted.
I did it 2 more times to be sure its not anything else. and same results.

PaintYourDragon, 2017-02-16:
Tested against both Nov 2016 Raspbian and latest Retropie. Buttons, keyboard and network all continue to function.

Confirm that /boot/retropie.cfg corresponds to your actual button/stick wiring, using Broadcom GPIO pin numbers, and that these are not shared with any other GPIO-attached hardware.

closed 2017-02-16 by PaintYourDragon

xillwillx, 2017-02-19:
i have 0 connections to any gpio pins, which pi are you testing on , im on a pi 2 with just a usb keyboard connected. 

i use win32imager to put either retropie image or raspian , boot up and everything is fine. i run 
> cd
curl -O https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/retrogame.sh
sudo bash retrogame.sh

and i choose 6 button and it asks me to reboot. once i reboot no network connection and no keyboard working.

PaintYourDragon, 2017-02-19:
Tested on v 1.1 and 1.2 Pi 2 hardware. USB keyboard, Ethernet and HDMI connected. Still can't reproduce. Anything else unique about the setup?

xillwillx, 2017-02-19:
just have the pi attached to hdmi , usb keyboard , cat5 for network, nothing on the gpio pins 
retropie-4.1-rpi2_rpi3.img on a sony sd card. its only after i run the script that everything fails after reboot. no network, no usb

xillwillx, 2017-02-22:
are you testing on pi2?

ladyada, 2017-02-22:
yes he tested on a pi 2

xillwillx, 2017-02-22:
would the code work on a model B rev 2.0 or would you have to change alot of code?

ladyada, 2017-02-22:
it should work on any pi