adafruit/Adafruit-Retrogame #41: Question: trying to use gpio 14 and 15?

jtbdevelopment, 2017-02-22:
Pi noob.  Trying to use retropie on raspberry pi 3 and leverage serial ports for two buttons but can't get them to register.  Is there something special I am missing.  I think I disabled serial console, bt uart etc but no luck.  Details here.

https://retropie.org.uk/forum/topic/8408/attempting-to-use-gpio-14-15-18-for-button-input

Any links that would help, articles seem to imply I should be able to do it if I'm not using serial port..

Thanks

ladyada, 2017-02-22:
plz dont use those pins - theres plenty of others! :)

closed 2017-02-22 by ladyada

jtbdevelopment, 2017-02-22:
Unfortunately using the rest (2 joysticks, 5 general buttons, 6 player buttons each, 2 short).  It seemed like it was feasible to use those pins if I was willing to forgo serial,bt which I am.  Is this wrong?

But perhaps not.  Reading your latest code earlier, it seems like I could build an extension kit with the mmc chip (or whatever it was called)?  I saw your notes on how to use the latest retro game with retropie in reader?

ladyada, 2017-02-22:
sure go for it! this code is what we offer only