adafruit/Adafruit-Retrogame #42: Add delay to key stroke

hiulit, 2017-03-21:
Is there any way to add a delay so that the key stroke "activates" after a certain period of time?

PaintYourDragon, 2017-03-21:
Afraid not. Closest thing is the "Vulcan nerve pinch" which activates when a button combo is held for a couple of seconds. Doesn't work for single buttons, but using an older version of the code (before settings files were added) it might be possible to hardcode this onto a single button. As long as just one is sufficient.

No plans to add this. It's never come up before, and the code is a terrifying spaghetti monster as it is.

closed 2017-03-21 by PaintYourDragon

hiulit, 2017-03-21:
That's even better! That's actually what I was looking for :)
Basically I'd like to add a delay for the combination "hotkey + another button" in Retropie.
How can I enable this "vulcan nerve pinch"?
Does this work with newer versions of Retrograme?

PaintYourDragon, 2017-03-21:
Oh! Well there you go. It does work in newer retrogame...in the settings file, follow the key code with multiple GPIO pin numbers instead of just one. See configs/retrogame.cfg.2button for an example.

The only "gotcha" is that you can have only ONE such combination.

hiulit, 2017-03-22:
Ok, I see what you say.
```
ESC      23 7 # Esc key = hold 'A'+'B' buttons (exit ROM)
```
Is that it?

But I don't see where I can set the time for the "vulcan nerve pinch".

Also, splitting hairs, could it be 3 or more buttons? :P

P.S. I'm okay with just one combination, thanks for the heads up!

PaintYourDragon, 2017-03-22:
Correct with the 'ESC 23 7'. Can be as many buttons as you want.

The time is hardcoded at 1.5 seconds. If you need to change that, currently the only option is to edit the source and recompile. 'vulcanTime' around line 114 (expressed in milliseconds).