adafruit/Adafruit-Retrogame #47: Remapping of select button w/ PiGrrl 2

DRCHRIS2, 2017-09-24:
A few folks have noted that on the PiGrrl 2 build the select button is getting mapped to [SPACE] currently, while also using this config in the instructions for the controller carries a default mapping of [SPACE] to toggle fast-forward... does not play nicely. Hitting select in any emulator will toggle fast-forward w/ the default config, no bueno.

Cf. https://retropie.org.uk/forum/topic/366/pigrrl-2-gamepad-select-button-acting-as-a-fast-forward-toggle-in-all-emulators/2

This is an easy fix that can be applied in the retroarch.cfg file included in _this_ repo that is being pulled down by the retrogame.sh script in the Raspberry-Pi-Installer-Scripts repo.

PaintYourDragon, 2017-09-24:
We changed the software installation method a few months ago. Rather than a ready-made SD image, we have folks download the current RetroPie release and then run PiTFT & retrogame installer scripts atop this.

When installed this way, EmulationStation will prompt for controller configuration when first run. With a handful of exceptions, ONLY these keys are assigned to input functions. input_hold_fast_forward, for example, is commented out by default in /opt/retropie/configs/all/retroarch.cfg. So, if installing the currently preferred way, the fast-forward situation shouldn't happen.

That said, there is still a little overlap in the default retroarch.cfg inputs and what's set up in most of the retrogame.cfg.* files, and better keys could probably be chosen. This could be done entirely in the retrogame.cfg.* files installed by the script, no modification to retroarch.cfg should be necessary.

Here's what appears to be the not-commented-out keys mentioned in retroarch.cfg. As long as retrogame.cfg.* steer clear of these, everything should be golden, I think. Yeah?
```
input_exit_emulator = "num1"
input_shader_next = "m"
input_shader_prev = "n"
input_rewind = "r"
input_menu_toggle = "z"
input_enable_hotkey = "num5"
input_reset = "alt"
```

DRCHRIS2, 2017-09-25:
Indeed, when you flash the latest RetroPie and run the retrogame script on top of it, and proceed to the EmulationStation mapping setup, the process has the PiGrrl2 Select button mapped to SPACE. I checked out the retroarch.cfg and indeed saw that the fast_forward options are commented out, but even when testing on a keyboard the SPACE button continues to throw the emulators into fast_forward mode.  Uncommenting those lines and mapping to another key fixes it. Therefore, another config file must be taking precedence when those lines are commented out and continuing to map Select to SPACE, right?

Any thoughts on what other config files may be doing this? It happens irrespective of the emulator launched. I took a look around for a good while and did not see anything, but I admit I didn't grep through the filesystem exhaustively.