adafruit/Adafruit-Retrogame #48: Error Installing Retrogame and Installing Retrogame.cfg

shanedertrain, 2017-10-01:
Hello,

I am having an issue with installing Retrogame. 

The initial download from Github flows smoothly, but when I attempt to run the script I encounter the following issue: 

[image]

What are the steps I should take in troubleshooting this?

PaintYourDragon, 2017-10-01:
Most likely Github was temporarily down or inaccessible. Try the script again. Or enter these commands (normally done in the script) and see what actual errors they're throwing:

curl -f -s -o /tmp/retrogame https://raw.githubusercontent.com/adafruit/Adafruit-Retrogame/master/retrogame

curl -f -s -o /boot/retrogame.cfg https://raw.githubusercontent.com/adafruit/Adafruit-Retrogame/master/configs/retrogame.cfg.2button

shanedertrain, 2017-10-01:
Attempt #2:

[image]
[image]

They do not seem to be returning anything.

PaintYourDragon, 2017-10-01:
I apologize. Leave the '-s' out of the curl commands (that makes it 'silent' and we can't see any error messages).

PaintYourDragon, 2017-10-01:
Try logging in as user 'pi' (not root) and use sudo when running the install script.

shanedertrain, 2017-10-01:
Looks like the script had to be run with the insecure, or -k command. 

[image]

Thanks a ton! 

I am curious, however, as to why this verification of the certificate fails.

shanedertrain, 2017-10-01:
I'm not sure if I should create another issue for this: 

Now it seems that the controls aren't working, even when shorting the pins to ground. 

I ran 'ps -ef | grep retrogame' and this is what returned: 

[image]

I also ran 'ps -e' and did not see retrogame in the list of running processes, so I'm assuming that the process is not running.

PaintYourDragon, 2017-10-01:
What version of RetroPie is this? And which Pi board?

Try running retrogame from the command line (don't background the process) and see if it reports anything.
sudo /usr/local/bin/retrogame

(Assuming, of course, you've gotten it to download and have moved the executable there. Install script should do this.)

shanedertrain, 2017-10-01:
I am using what I am assuming is the latest version of Retropie, having downloaded the image from their website last week, and the Pi 3 model B. 

sudo /usr/local/bin/retrogame returns "command not found". Did the install script download fail too?

PaintYourDragon, 2017-10-01:
The curl command puts it in /tmp. Move it to /usr/local/bin.
Though, for the sake of testing, you can try just running it from /tmp, no biggie.

shanedertrain, 2017-10-01:
Looks like there's nothing in /tmp nor /usr/local/bin.

PaintYourDragon, 2017-10-01:
sudo chmod 777 /usr/local/bin/retrogame
sudo /usr/local/bin/retrogame

shanedertrain, 2017-10-01:
That did the trick. Thanks a ton!

PaintYourDragon, 2017-10-05:
Try start+select rather than A+B.